arduino/arduino-pro-ide #13: Cannot fold functions with comments

naamah75, 2019-10-19:
Cannot fold functions that have comments inside.

The functions is folded only until the comment, then the remaining part of the functions remain unfolded.

[image]

[image]

 - OS: Deepin
 - Version 15.11

kittaakos, 2019-10-23:
Thank you for logging, this is a language server issue.

spoenemann, 2020-01-13:
The language server does not define any folding. Maybe it's buried in the TextMate grammar, or some default folding of Monaco?

kittaakos, 2020-01-20:
> language server does not define any folding

I am confused, what do you mean?

[`textDocument/foldingRange`](https://microsoft.github.io/language-server-protocol/specifications/specification-3-14/#textDocument_foldingRange):
 > The folding range request is sent from the client to the server to return all folding ranges found in a given text document.

spoenemann, 2020-01-20:
Yes, but our current language server doesn't support that request.

kittaakos, 2021-02-11:
It works now.

closed 2021-02-11 by kittaakos