arduino/arduino-pro-ide #15: Find in files do not work

dronecz, 2019-10-19:
**Describe the bug**
When I have multiple tabs open in my project,  search in files do not work. I´m able to find what I´m looking in one tab, but not in all tabs. 

[image]
[image]

 - OS: Windows 7 x64 Pro SP1; Windows 10 (1903)

kittaakos, 2019-10-23:
@dronecz, where is your `test.h` and the two opened `sketch_october19c.ino` files on the file system?

dronecz, 2019-10-23:
These were just temporary files which I created in that time, but I think test.was at my Desktop. Files from dark screenshot are stored in my Dropbox folder.

spoenemann, 2020-01-13:
I created an issue here: https://github.com/eclipse-theia/theia/issues/6867

per1234, 2021-02-19:
Hi @dronecz . Thanks so much for taking the time to report this!

This should now be working in [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4).

If you try it out and find that the problem is still occurring, feel free to comment here and we'll investigate.

closed 2021-02-19 by per1234