arduino/arduino-pro-ide #16: Use of non-standard macOS short cuts

unknowndomain, 2019-10-19:
**Describe the bug**
The short cuts used are non standard.

**Expected behavior**
Examples:
- Command + W = Close Window
- Command + , = Preferences
- Command + Shift + S = Save As

sbhklr, 2019-10-20:
**More examples:**
- Command + O = Open File
- Command + "+" = Zoom in 
- Command + N = New File

tigoe, 2019-11-06:
+1 to this issue. OS standard keyboard shortcuts are well-known, and should be respected. There are even some that are cross-OS that this system does not respect.  I regard this as a bug.

The system should also respect as many of the original Arduino app keyboard shortcuts as possible, IMO, per issue #17

jankeromnes, 2019-11-25:
FYI, I've updated Theia to use `Command + , = Preferences` by default on Mac OS: https://github.com/eclipse-theia/theia/pull/6620

In general, any keybindings that don't seem "standard" (i.e. they're different than VS Code) should be fixed in Theia.

On the other hand, any Arduino-specific keybindings (e.g. `Command + U = Upload`) can be defined in this project, by implementing a `KeybindingContribution` class with a `registerKeybindings(keybindings: KeybindingRegistry): void` method.

jankeromnes, 2019-11-26:
> Command + W = Close Window

In VS Code, this is different for each OS:

- Mac OS:
  - Command + W = Close Editor
  - Shift + Command + W = Close Window
- Windows:
  - Ctrl + F4 = Close Editor
  - Ctrl + W = Close Window

Would these keybindings be acceptable in Arduino Pro too?

jankeromnes, 2019-12-03:
Update: the majority of keybindings have been fixed upstream in Theia, so we should update Theia here whenever convenient.

Another thing that remains to be done here is Arduino-specific bindings like Cmd + U = Upload.

jankeromnes, 2019-12-12:
With the latest Theia update and v0.0.3 release, this is now done. Hooray! Thanks all for your input.

Going forward, if you feel like a native-like keybinding is missing, please open a dedicated issue.

Note that there is already #17 about supporting legacy Arduino IDE shortcuts like <kbd>Cmd</kbd> + <kbd>U</kbd> = `Upload`.

closed 2019-12-12 by jankeromnes