arduino/arduino-pro-ide #17: Support standard Arduino app shortcuts

unknowndomain, 2019-10-19:
**Is your feature request related to a problem? Please describe.**
It will become frustrating that shortcuts like Command + U for upload don't work.

**Describe the solution you'd like**
Add all the exisitng arduino app shortcuts.

**Describe alternatives you've considered**
Adding shortcuts through System Preferences > Keyboard but there are very few menu items.

kittaakos, 2019-10-23:
Related #16

spoenemann, 2020-01-13:
I added two shortcuts:

 * Verify: `ctrl+alt+V` (`cmd+alt+V` on Mac)
 * Upload: `ctrl+alt+U` (`cmd+alt+U` on Mac)
    In the Classic IDE it was `ctrl/cmd+U`, but this conflicts with the [VS Code keybinding](https://code.visualstudio.com/docs/getstarted/keybindings#_default-keyboard-shortcuts) for "Undo last cursor operation".

Please open another issue if you're missing more shortcuts.

closed 2020-01-13 by spoenemann