arduino/arduino-pro-ide #20: macOS: Can't select port

rei-vilo, 2019-10-19:
**Describe the bug**
The dialogue box allows to select the board but not the port.

**To Reproduce**
Steps to reproduce the behavior:

1. Install the main Arduino boards in the IDE
2. Connect the board to the computer
3. Select other board
4. The board can be selected but not the port.

**Expected behavior**
Port is listed and can be selected.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 10.15 _Catalina_
 - Version:  [image]

**Additional context**
Similar to #8 but on macOS and for missing port only.

mihmig, 2019-10-20:
Some problem on Windows version...

squio, 2019-10-20:
Same here: 

- Mac OS Mojave
- Arduino-PoC  0.0.1 (0.0.1.20191018.1)

Original Genuine Arduino Uno attached to USB (works in traditional Arduino IDE)
No port discovered.

lemio, 2019-10-20:
Same here:

- MacOS Catalina
- 0.0.1 (0.0.1.20191018.1)
- arduino-electron 0.0.1

UNO and mega don't work, while working on the normal IDE

nikilase, 2019-10-20:
Also same problem on Windows 10 18362
It can't find my Uno WiFi Rev2 in the new IDE(No Port Discovered) but it can be found in the old IDE.
On my Laptop the Arduino can be found in the new IDE but I have #38

BaldEagleX02, 2019-10-21:
Same here with MacOS Catalina, arduino-pro-ide 0.0.1 (0.0.1.20191018.1) and arduino-electron 0.0.1

rsora, 2019-10-21:
Hi there, Thank you all for the detailed reports!

In the meantime, can you please check if the snapshot builds attached here https://github.com/arduino/arduino-pro-ide/issues/51#issuecomment-544560707 solve your detection problem?

Thanks!

mastrolinux, 2019-10-25:
Should be fixed by https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview

rei-vilo, 2019-10-25:
On my case, the issue is fixed with Alpha 0.0.2 on macOS.

closed 2019-10-25 by rei-vilo