arduino/arduino-pro-ide #21: macOS: Missing fields in boards manager list

rei-vilo, 2019-10-19:
**Describe the bug**
Missing fields in boards manager list

**To Reproduce**
Steps to reproduce the behavior:
1. Open Tools > Boards manager

**Expected behavior**
Fully populated list with names and providers.
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
OS: macOS 10.15 Catalina
Version: [image]

**Additional context**
Similar to #10 but on macOS.

hex705, 2019-10-20:
same issue osx 10.13.6 high sierra

mdxtinkernick, 2019-10-20:
Similar issue, but names are not populated until Arduino brand boards are reached, and all boards after that display name entry
[image]

rei-vilo, 2019-10-25:
The issue is still there on Alpha 0.0.2.

mastrolinux, 2019-10-28:
Adding more context:
Extra info are not displayed if a compatible core is not installed.

rei-vilo, 2019-10-28:
But Arduino UNO board is not available after installation "Arduino AVR boards". See #69

kittaakos, 2019-11-12:
How can I reproduce it? Do I have to use an additional board manager URL or do I have to install a core manually? Thanks!

kittaakos, 2019-11-13:
I was unable to reproduce this with the following list of 3rd party core support:
```
'https://downloads.arduino.cc/packages/package_index.json',
"https://adafruit.github.io/arduino-board-index/package_adafruit_index.json",
"https://raw.githubusercontent.com/akafugu/akafugu_core/master/package_akafugu_index.json",
"https://raw.githubusercontent.com/AloriumTechnology/Arduino_Boards/master/package_aloriumtech_index.json",
"https://github.com/Ameba8195/Arduino/raw/master/release/package_realtek.com_ameba_index.json",
"https://raw.githubusercontent.com/arachnidlabs/arachnidlabs-boards/master/package_arachnidlabs.com_boards_index.json",
"https://ardhat.github.io/ardhat-board-support/arduino/package_ardhat_index.json",
"https://arduboy.github.io/board-support/package_arduboy_index.json",
"https://per1234.github.io/Ariadne-Bootloader/package_codebendercc_ariadne-bootloader_index.json",
"https://raw.githubusercontent.com/ioteamit/smarteverything-core/master/package_arrow_index.json",
"https://mesom.de/atflash/package_atflash_index.json",
"https://thomasonw.github.io/ATmegaxxM1-C1/package_thomasonw_ATmegaxxM1-C1_index.json",
"https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json",
"https://raw.githubusercontent.com/avandalen/SAM15x15/master/package_avdweb_nl_index.json",
"https://zevero.github.io/avr_boot/package_zevero_avr_boot_index.json",
"https://raw.githubusercontent.com/VSChina/azureiotdevkit_tools/master/package_azureboard_index.json",
"https://raw.githubusercontent.com/carlosefr/atmega/master/package_carlosefr_atmega_index.json",
"https://raw.githubusercontent.com/oshlab/Breadboard-Arduino/master/avr/boardsmanager/package_oshlab_breadboard_index.json",
"https://resources.canique.com/ide/package_canique_index.json",
"https://github.com/chipKIT32/chipKIT-core/raw/master/package_chipkit_index.json",
"https://raw.githubusercontent.com/mikaelpatel/Cosa/master/package_cosa_index.json",
"https://raw.githubusercontent.com/CytronTechnologies/Cytron-Arduino-URL/master/package_cytron_index.json",
"https://raw.githubusercontent.com/DFRobot/DFRobotDuinoBoard/master/package_dfrobot_index.json",
"https://raw.githubusercontent.com/DFRobot/DFRobotDuinoBoard/master/package_dfrobot_iot_mainboard.json",
"https://git.oschina.net/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json",
"https://raw.githubusercontent.com/MauricioJancic/Elemon/master/package_Elemon_index.json",
"https://raw.githubusercontent.com/ElektorLabs/arduino/master/package_elektor-labs.com_ide-1.6.6_index.json",
"https://raw.githubusercontent.com/ElektorLabs/arduino/master/package_elektor-labs.com_ide-1.6.5_index.json",
"https://engimusing.github.io/arduinoIDE/package_engimusing_modules_index.json",
"https://arduino.esp8266.com/stable/package_esp8266com_index.json",
"https://dl.espressif.com/dl/package_esp32_index.json",
"https://raw.githubusercontent.com/FemtoCow/ATTinyCore/master/Downloads/package_femtocow_attiny_index.json",
"https://raw.githubusercontent.com/harbaum/ftduino/master/package_ftduino_index.json",
"https://raw.githubusercontent.com/feilipu/feilipu.github.io/master/package_goldilocks_index.json",
"https://github.com/Infineon/Assets/releases/download/current/package_infineon_index.json",
"https://github.com/IntoRobot/IntoRobotPackages-ArduinoIDE/releases/download/1.0.0/package_intorobot_index.json",
"https://raw.githubusercontent.com/ioteamit/ioteam-arduino-core/master/package_ioteam_index.json",
"https://raw.githubusercontent.com/udif/ITEADSW_Iteaduino-Lite-HSP/master/package/package_iteaduino_lite_index.json",
"https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json",
"https://raw.githubusercontent.com/eightdog/laika_arduino/master/IDE_Board_Manager/package_project_laika.com_index.json",
"https://macchina.cc/package_macchina_index.json",
"https://www.mattairtech.com/software/arduino/package_MattairTech_index.json",
"https://raw.githubusercontent.com/MaximIntegratedMicros/arduino-collateral/master/package_maxim_index.json",
"https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json",
"https://mcudude.github.io/MicroCore/package_MCUdude_MicroCore_index.json",
"https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json",
"https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json",
"https://lowpowerlab.github.io/MoteinoCore/package_LowPowerLab_index.json",
"https://nekuneko.github.io/arduino-board-index/package_nekuneko_index.json",
"https://raw.githubusercontent.com/NicoHood/HoodLoader2/master/package_NicoHood_HoodLoader2_index.json",
"https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json",
"https://raw.githubusercontent.com/OLIMEX/Arduino_configurations/master/AVR/package_olimex_avr_index.json",
"https://raw.githubusercontent.com/OLIMEX/Arduino_configurations/master/PIC/package_olimex_pic_index.json",
"https://github.com/chipKIT32/chipKIT-core/raw/master/package_chipkit_index.json",
"https://raw.githubusercontent.com/OLIMEX/Arduino_configurations/master/STM/package_olimex_stm_index.json",
"https://raw.githubusercontent.com/ThamesValleyReprapUserGroup/Beta-TVRRUG-Mendel90/master/Added-Documents/OMC/package_omc_index.json",
"https://openpanzerproject.github.io/OpenPanzerBoards/package_openpanzer_index.json",
"https://raw.githubusercontent.com/geolink/opentracker-arduino-board/master/package_opentracker_index.json",
[... cut ...]
"https://per1234.github.io/wirino/package_per1234_wirino_index.json",
"https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json",
"https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json",
"https://github.com/XMegaForArduino/IDE/raw/master/package_XMegaForArduino_index.json",
"https://raw.githubusercontent.com/sanu-krishnan/ebot-arduino-core/master/package_ebots.cc_index.json"
```

@rei-vilo, @hex705, do you have `arduino-cli` on the `PATH`? If so, what version does it have?

kittaakos, 2019-11-21:
Can I get some feedback from anyone, please? How can I reproduce the defect?

CC: @mastrolinux

hex705, 2019-11-21:
sorry for delay - here is my context. 

i am on osx 10.13.6
arduino-cli is on my path  -- version 0.6.0
downloaded arduino-pro 2.0 with all other versions of pro deleted -- not sure what a clean install procedure would be -- from here:
https://github.com/arduino/arduino-pro-ide/releases

reading the thread I realized -- 

i have a version of Arduino 1.8.10 also installed on my machine with following url based boards added on:
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json,
http://arduino.esp8266.com/stable/package_esp8266com_index.json,
https://raw.githubusercontent.com/Lauszus/Sanguino/master/package_lauszus_sanguino_index.json

I also have a manually added board in -->   Arduino/hardware/ folder 
When I open arduino-pro-ude in pro mode and open board manager I get same behaviour as @mdxtinkernick  above (oct 20).

The boards that do not have dev info in the pro-ide board menu seem to match the manual add-on (hardware folder) and the URLs I have added to my regular Arduino install.    When I scroll down I do get to Arduino boards with developer and other info included.

I am guessing that there is a some incomplete cross-over of platforms ?

hope this helps.

[image]

masci, 2019-11-22:
> > in --> Arduino/hardware/ folder

> Did you mean ~/Library/Arduino15/packages/arduino/hardware?

Hardware can live within the sketchbook folder, so for OSX that would be `~/Documents/Arduino/hardware`

kittaakos, 2019-11-22:
Here are the steps to reproduce the problem with the CLI only. I am not sure if this is an IDE problem, maybe  the way we update the index and run the search from the IDE is incorrect.

 1. Wipe the previous state if any:
```
rm -rf ~/Library/Arduino15
```

 2. Check the CLI version:
```
./arduino-cli version
```

 3. Update index with a 3rd party core:
```
./arduino-cli core update-index --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json
```

 4. Run a wildcard seach for all cores:
```
./arduino-cli core search --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json
```

 5. Install the `arbotix:avr` 3rd party core:
```
./arduino-cli core install arbotix:avr --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json
```

 6. Update the index without the URL of the previously installed 3rd party core:
```
./arduino-cli core update-index
```

 7. Run a new platform search with the additional core URL; you can see the 3rd part core and its `Name`:
```
./arduino-cli core search --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json
```

 8. Run another platform search without the additional core URL; you cannot see the `Name` of the installed 3rd party core:
```
./arduino-cli core search
```

All in one (you can see the final search does not contain the `Name` of the installed 3rd party core):
```
rm -rf ~/Library/Arduino15 \
&& ./arduino-cli version \
&& ./arduino-cli core update-index --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json \
&& ./arduino-cli core search --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json \
&& ./arduino-cli core install arbotix:avr --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json \
&& ./arduino-cli core update-index \
&& ./arduino-cli core search --additional-urls https://tician.github.io/arbotix/package_trossen_arbotix_index.json \
&& ./arduino-cli core search
```

End of the output:
```
[...]

ID                   Version   Name                                             
Arrow:samd           2.1.0     Arrow Boards                                     
Intel:arc32          2.0.4     Intel Curie Boards                               
Intel:i586           1.6.7+1.0 Intel i586 Boards                                
Intel:i686           1.6.7+1.0 Intel i686 Boards                                
Microsoft:win10      1.1.2     Windows 10 Iot Core                              
arbotix:avr          0.0.3     Arbotix AVR Boards                               
arduino:avr          1.8.1     Arduino AVR Boards                               
arduino:mbed         1.1.2     Arduino nRF528x Boards (Mbed OS)                 
arduino:megaavr      1.8.5     Arduino megaAVR Boards                           
arduino:nrf52        1.0.2     Arduino nRF52 Boards                             
arduino:sam          1.6.12    Arduino SAM Boards (32-bits ARM Cortex-M3)       
arduino:samd         1.8.4     Arduino SAMD Boards (32-bits ARM Cortex-M0+)     
arduino:samd_beta    1.6.25    Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)
atmel-avr-xminis:avr 0.6.0     Atmel AVR Xplained-minis                         
emoro:avr            3.2.2     EMORO 2560                                       
industruino:samd     1.0.1     Industruino SAMD Boards (32-bits ARM Cortex-M0+) 
littleBits:avr       1.0.0     littleBits Arduino AVR Modules                   

ID                   Version   Name                                             
Arrow:samd           2.1.0     Arrow Boards                                     
Intel:arc32          2.0.4     Intel Curie Boards                               
Intel:i586           1.6.7+1.0 Intel i586 Boards                                
Intel:i686           1.6.7+1.0 Intel i686 Boards                                
Microsoft:win10      1.1.2     Windows 10 Iot Core                              
arbotix:avr          0.0.3                              <------ MISSING NAME :(
arduino:avr          1.8.1     Arduino AVR Boards                               
arduino:mbed         1.1.2     Arduino nRF528x Boards (Mbed OS)                 
arduino:megaavr      1.8.5     Arduino megaAVR Boards                           
arduino:nrf52        1.0.2     Arduino nRF52 Boards                             
arduino:sam          1.6.12    Arduino SAM Boards (32-bits ARM Cortex-M3)       
arduino:samd         1.8.4     Arduino SAMD Boards (32-bits ARM Cortex-M0+)     
arduino:samd_beta    1.6.25    Arduino SAMD Beta Boards (32-bits ARM Cortex-M0+)
atmel-avr-xminis:avr 0.6.0     Atmel AVR Xplained-minis                         
emoro:avr            3.2.2     EMORO 2560                                       
industruino:samd     1.0.1     Industruino SAMD Boards (32-bits ARM Cortex-M0+) 
littleBits:avr       1.0.0     littleBits Arduino AVR Modules                   

```

hex705, 2019-11-22:
HI All,

to clarify -- I meant ::

/users/myCOmputerName/Documents/Arduino/hardware

which has my manually added board -- in the hardware folder:
/custom/avr/boards.txt
/custom/avr/bootloaders

boards.txt has this content:
unorestore.name=UNO 16U2 Restore Firmware
unorestore.bootloader.tool=arduino:avrdude
unorestore.bootloader.low_fuses=0xff
unorestore.bootloader.high_fuses=0xd9
unorestore.bootloader.extended_fuses=0xf4
unorestore.bootloader.file=Arduino-usbserial-atmega16u2-Uno-Rev3.hex
unorestore.bootloader.unlock_bits=0x3F
unorestore.bootloader.lock_bits=0x0F
unorestore.build.mcu=atmega16u2
unorestore.build.f_cpu=16000000L
unorestore.build.board=uno.build.board=AVR_UNO
unorestore.build.core=arduino:arduino
unorestore.build.variant=standard

bootloader is  Arduino-usbserial-atmega16u2-Uno-Rev3.hex

which i got off the internet somewhere to update the 16u2 chip on a board at some point in my past.

hope that helps.

s

kittaakos, 2019-12-03:
Update issue in the CLI: https://github.com/arduino/arduino-cli/issues/506

jankeromnes, 2019-12-12:
v0.0.3 has been released, moving this issue to v0.0.4 milestone.

kittaakos, 2020-08-26:
This issue has been [fixed in the CLI](https://github.com/arduino/arduino-cli/pull/556).

closed 2020-08-26 by kittaakos