arduino/arduino-pro-ide #22: Libraries are not being recognised

mevolkan, 2019-10-19:
**Describe the bug**
I have installed all the required libraries to run my sketch but the editor is showing errors in the markup and showing the libraries are not found

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version  1903

squio, 2019-10-20:
Same problem:

- Mac OS Mojave
- Arduino-PoC 0.0.1 (0.0.1.20191018.1)

Library is in include path according to settings.

[image]

spoenemann, 2020-01-13:
> Library is in include path according to settings.

@squio which settings?

spoenemann, 2020-01-14:
The Adafruit and RTC libraries should work with the next release v0.0.4.

closed 2020-01-14 by spoenemann