arduino/arduino-pro-ide #3: Incorrect string literal highlighting in dark theme with "%" character

spacegaier, 2019-10-19:
**Describe the bug**
"%" is in red font color instead of the orange that the rest of the string literal uses.

**To Reproduce**
Steps to reproduce the behavior:
1. Start IDE with dark theme.
2. New INO file with source code: const char* string = "string%withPercent";

**Expected behavior**
All orange font for string literal.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Win 10
 - Version 0.0.1

svenefftinge, 2019-11-28:
I cannot reproduce this on OSX with the next milestone (v0.0.3).
[image]

spacegaier, 2020-01-07:
It seems to depend on the actual string literal. Your "foo%test" also works fine for me, but my original test value still has the wrong highlight in v0.0.3 as does a simple value like "a%b". "a%a" however is weirdly fine.
[image]

spoenemann, 2020-01-10:
That's because the highlighting supports _format specifiers_ used by [printf](http://www.cplusplus.com/reference/cstdio/printf/) and other standard functions. It does not check whether the string is actually used within a call to `printf`, but that's a limitation we cannot overcome.

closed 2020-01-10 by spoenemann