arduino/arduino-pro-ide #4: IDE launches -  the editor opens, but nothing is happening but the "loading" circle in the middle

talofer99, 2019-10-19:
IDE launches -  the editor opens, but nothing is happening but the "loading" circle in the middle.
I left it running guessing it might be downloading or doing something in the background for more then 20 minutes - nothing happen still.
Tried all the options on the menu - no response from none of them with the except of the toggle developer tool 
I even tried running it as administrator as well same thing. 

Windows 10 pro 64bit 
Have no internet connection issues, run cli on the computer in the past with no issues.

spacegaier, 2019-10-19:
I had the same issue at first (also Win 10). I then closed and started the app again (one or two times) and from then on it worked fine.

talofer99, 2019-10-19:
@spacegaier tried that few times. 
The next thing will be to restart the computer ;)

nikilase, 2019-10-19:
IDE started up ok for me, well except #9 , but when I tried running it as an admin it did the endless loading 4 times in a row, then it went back to starting ok. 
Then I tried running it as an admin again and the endless loading came back and went away after the second restart. 
Oddly enough after the third try running it as an admin it started up ok, but seemed to have lost the config I made before i.e. the Advanced Mode toggle and the active views.

talofer99, 2019-10-19:
I will add that now I finally noticed the error on the console .

preference-service.ts:264 Uncaught (in promise) Error: Unable to write to Default Settings.
    at PreferenceServiceImpl.<anonymous> (preference-service.ts:264)
    at step (preference-service.ts:15)
    at Object.next (preference-service.ts:15)
    at fulfilled (preference-service.ts:15)

I get to this point ..... 

    throw new Error(`Unable to write to ${PreferenceScope.getScopeNames(resolvedScope)[0]} Settings.`);

kittaakos, 2019-10-23:
@talofer99, do you see anything else in the logs? Where is your executable, does the path contain a space? It is a known issue: https://github.com/arduino/arduino-pro-ide/issues/7#issuecomment-545528214

If you do not have space in the path, please read further 😊

What does the app log if you start it from a terminal? If the app starts, it should be possible to open the developer tools with the <kbd>Ctrl/Cmd</kbd>+<kbd>Alt</kbd>+<kbd>I</kbd> keybinding even if the application menu does not load; what do you see in the console:

Here is a screenshot from my console inside the dev tools:
[image]

talofer99, 2019-10-23:
@kittaakos  thank you very much. 
I did not even think of running it from a command line. 
It was a mix of having space and old version of the CLI 
Got it to work - thanks again :)

kittaakos, 2019-10-24:
Thank you for your feedback, @talofer99.

This item is tracked under #9.

mastrolinux, 2019-10-28:
Should be fixed in https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview

closed 2020-01-13 by spoenemann