arduino/arduino-pro-ide #6: Failed to start ide clang not found. - Ubuntu

suryamurugan, 2019-10-19:
Failed to start ide clang not found.

Steps to reproduce the behavior:
1. Loading the ide on ubuntu 16

 - OS: Ubuntu
 - Version 16.04

naamah75, 2019-10-19:
Same issue here on Deepin 15.11

mastrolinux, 2019-10-19:
This can be solved by:

`snap install clans --classic`

Can you please try it out and let us know?

naamah75, 2019-10-19:
I' sorry but I cannot find the snap "clans".

naamah75, 2019-10-19:
Installed snap "clangd", the problem seem to be solved.

mastrolinux, 2019-10-19:
Update the comment, closing this for now.

closed 2019-10-19 by mastrolinux

djsegfault, 2019-10-20:
As per #31 I tried "sudo apt-get install clang-tools-5.0" and that didn't fix it.  I also tried "sudo snap install clangd --classic" and that also didn't fix it. I got the same error

jpeterse, 2019-10-20:
Getting the same error, even though Clang is already installed:

$ clang --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

$ locate clangd
/usr/bin/clangd-8
/usr/lib/llvm-8/bin/clangd

notkevinjohn, 2019-10-21:
I got around this with the following steps on Ubuntu 19.04:

Install the clang tool package:
$ sudo apt-get install clang-tools-8

Specify the appropriate alternative for clangd
$ sudo update-alternatives --install /usr/bin/clangd clangd /usr/bin/clangd-8 100

These were instructions I found on clang's website:
https://clang.llvm.org/extra/clangd/Installation.html

jpeterse, 2019-10-21:
I already had clang-tools installed, but didn't have a link from /usr/bin/clangd-8 to /usr/bin/clangd.
Adding the alternate option creates this link, and Arduino editor is now locating clangd.

cwfitzgerald, 2020-01-08:
I set the `cpp.clangdExecutable` settings flag to `clangd-8` as I had clang-tools-8 installed, but it still reports a failure to start clangd.