arduino/arduino-pro-ide #7: Only "Help" button after starting up on windows 10.

B45i, 2019-10-19:
**Only a help button in the title bar which contains toggle developer options**
After I extracted the zip file and execute the `Arduino-PoC.exe`, there is nothing on the screen,
Toggle developer option is available although it says developer tools is disconnected.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Winsows 10 pro
 - Version : Version	10.0.18362 Build 18362

kittaakos, 2019-10-23:
@B45i, thank you for the detailed bug report. It helped a lot 👍

As a quick workaround; please remove the spaces from the `Arduino IDE Pro` folder name. The whitespaces in the path cause the issue. 😕

kittaakos, 2019-12-03:
We have fixed the issue in `0.0.2`. I am closing it.

Please reopen if required.

closed 2019-12-03 by kittaakos