arduino/arduino-pro-ide #8: Can't Select the Board and Port

rluberto, 2019-10-19:
The IDE opens and works properly however it is unable to discover the board and port.
Note: I installed the boards in the IDE however the boards I installed are still grayed out when trying to select a board.

Steps to reproduce the behavior:
1. Install the main Arduino boards in the IDE
2 Plug in board to computer
3. Select other board
4. No boards or ports are discovered

**Expected behavior**
Board or port is discovered

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro 64 bit
 - Version 10.0.18362

OMGdaDPS, 2019-10-20:
same here. Able to install the needed stuff through the board manager, but the IDE doesnt find any ports. I am on Ubuntu 19.10, and have added myself to the dialout group of course

rsora, 2019-10-21:
Hi there, thanks for your report!

Can you please check if the snapshot builds attached here https://github.com/arduino/arduino-pro-ide/issues/54#issuecomment-544558918 solve your detection problem?

Thanks!

rluberto, 2019-10-21:
Yes that solved the problem!

closed 2019-10-21 by rluberto