arduino/arduino-pro-ide #9: Board Manager not showing any Board on Windows 10

nikilase, 2019-10-19:
I can't find any board in the board manager. When I search "uno" or anything else I still don't get any board. 
The IDE has following error message when I start it:
> Editor support may be inaccurate because the core for the board arduino:avr:uno is not installed. Use the Boards Manager to install it.

In addition, I also can't find any library in the library manager. 

[image]

**Desktop**
 - OS: Windows 10
 - Version 0.0.1-alpha.preview-windows

nikilase, 2019-10-19:
Running it with admin privileges did not fix this.

nikilase, 2019-10-19:
I unzipped the alpha on my other PC too, this time on the Desktop as opposed to my Program Files, and the Board Manager also showed no entry. But this time I got another error message: 

> Could not start editor support.
Downloading missing tool builtin:ctags@5.8-arduino11...
0 / 116455 0.00% builtin:ctags@5.8-arduino11 downloaded
Installing builtin:ctags@5.8-arduino11...
installing builtin:ctags@5.8-arduino11 tool: installing tool builtin:ctags@5.8-arduino11: moving extracted archive to destination dir: rename C:\Users\nikla\AppData\Local\Arduino15\tmp\package-093549799\ctags-5.8-arduino11 C:\Users\nikla\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11: Access is denied.
Error initializing package manager: installing builtin:ctags@5.8-arduino11 tool: installing tool builtin:ctags@5.8-arduino11: moving extracted archive to destination dir: rename C:\Users\nikla\AppData\Local\Arduino15\tmp\package-093549799\ctags-5.8-arduino11 C:\Users\nikla\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11: Access is denied.

sworteu, 2019-10-19:
Having the same issue, i can't get any boards in the boards manager even after search entry. On windows 10.

nikilase, 2019-10-19:
Funny enough on the same laptop dual booted to Ubuntu 18.04 LTS I could install the program and just needed to start, close and start it and I was able to install libraries and boards.

Fede85, 2019-10-20:
I had a smilar issue (IDE unable to load the Board Manager) and running the executable by command line I figured out that it was failing to load the `arduino-cli` because the path contained a space ("`C:\Program Files`"). After moving to a different path, without spaces in the directory names, the IDE started and I was able to install a core.

I also noticed that the output box was displayed after "fixing" the path. Probably the "output" box appear only if the `arduino-cli` is loaded.

[image]

nikilase, 2019-10-20:
Crazy, removing all spaces fixed the issue. Can this be fixed by the Arduino Team or is this more on the level of user error? Either way it would be nice if they would add a little warning in their docs about this issue.

Fede85, 2019-10-20:
I think that is definetly something that should be fixed in the pro IDE. The user should be allowed to use paths containig spaces.

Since the core component of the pro IDE is the `arduino-cli` the frontend shall notify the user if it's not able to load it.

I'd suggest you to change the title of the issue with something more related with paths and spaces. This will help the Arduino team to sort out the issues more easily.

nikilase, 2019-10-20:
Is this title better, I'm not a native speaker so I'm not sure if the wording is perfect.

kittaakos, 2019-10-23:
> Crazy, removing all spaces fixed the issue.

👍 

Yes, this is unfortunately a known issue: https://github.com/arduino/arduino-pro-ide/issues/7#issuecomment-545528214

kittaakos, 2019-12-03:
We have fixed the issue in `0.0.2`. I am closing it.

Please reopen if required.

closed 2019-12-03 by kittaakos