arduino/arduino-pro-ide #23: Project with multiple tabs -- only main file opens.

hex705, 2019-10-20:
**Describe the bug**
When opening a file with multiple tabs, only the main file opens.
All tabs can be opened manually one at a time  -- but this is a pain if there many / more than 3.
_Compile fails if additional tabs not manually opened._ 

**To Reproduce**
Go to file manager (side bar, nice), 
Select any Arduino project with more than one tab 
Double click the file that matches folder name.  
Only that file opens
Additional files stay closed and need to be opened manually.

**Expected behavior**
As with old IDE -- expect all tabs/files to open.

**Desktop (please complete the following information):**
Mac OSX 10.13.6
High Sierra

mastrolinux, 2019-10-27:
This is only happening in pro mode, and pro users are expected to open only the files they need. We cannot open all the files at the same time for huge projects, so this is intended behavior.

closed 2019-10-27 by mastrolinux

per1234, 2020-03-16:
>This is only happening in pro mode

@mastrolinux was there a regression? Using version 0.0.5 in non-advanced mode, when I open a file of a multi-file sketch, only the file I selected is opened.