arduino/arduino-pro-ide #25: Feature request for the IDE

jojobudiman, 2019-10-20:
This is a feature request but I would love to this IDE could support more extensions, user-friendly and somehow support GitLens such as VSCode. Live share is a good option for monitoring and teaching people who are new to the IDE. File icons and fonts would be appreciated so we could develop in the IDE comfortably. Because this is just an alpha version, there's still a long way to go. Just giving good luck to the Arduino team and if the team could implement these features I mentioned for the IDE. It would be great! 😊

Don't forget to fix the user interface font and simplify the library description (would be nice if there were library icons 😃)
[image]

spoenemann, 2020-01-13:
> fix the user interface font

What's wrong with the font?

> add keyboard shortcuts

Done: #17

> simplify the library description (would be nice if there were library icons 😃)

Related: #21

jojobudiman, 2020-01-13:
> What's wrong with the font?
Only the board manager font though if it's unnecessary 🤔

I'll close the issue since enhancements were made.

closed 2020-01-13 by jojobudiman