arduino/arduino-pro-ide #27: Not able to connect to Arduino (SAMD) boards on mac Catalina

torenrl, 2019-10-20:
**Describe the bug**
Even though the SAMD board package is installed and the correct board is connected, no active port is found by the board manager. I am using macOS Catalina, and I have tested with Genuino Zero and Arduino WiFi 1010.

**To Reproduce**
Steps to reproduce the behavior:
1. Arduino-PoC on Catalina
2. Select Arduino WiFi 1010 or Genuino Zero from the board manager and connect the board
3. "NO PORTS DISCOVERED"

**Expected behavior**
I expected to find the Arduino on the port used, to be able to upload code

**Screenshots**
-

**Desktop (please complete the following information):**
 - OS: macOS
 - Version Catalina

torenrl, 2019-10-20:
Same as #20.

rsora, 2019-10-21:
Hi @torenrl , thanks for your report!

Can you please check if the ubuntu snapshot build attached here https://github.com/arduino/arduino-pro-ide/issues/54#issuecomment-544558918 solves your problem?

Thanks!

torenrl, 2019-10-22:
This works. Thank you.

closed 2019-10-22 by torenrl