arduino/arduino-pro-ide #29: tabs do not have the closing "x"

DaRaymond1, 2019-10-20:
Tabs inside the IDE do not have the closing "x" and when right clicking and selecting close nothing happens. as far as i can tell there are no ways of closing tabs other than closing the entire IDE.

Fede85, 2019-10-20:
The "Advanced mode" allows you to close the files. The issue seems related only to the "Classic mode"

MrRSquared, 2019-10-20:
Even if you right click on the tab and select close (or any variation, close to the right etc. ), nothing closes.  Maximize and minimize work.  In the dark version of pro mode, it is difficult to see the xs, but they are there and work as expected.

mastrolinux, 2019-10-28:
This is the intended behavior. It is the same in the Arduino classic IDE. Closing the issue.

closed 2019-10-28 by mastrolinux