arduino/arduino-pro-ide #30: An unexpected error occurred when locating the Git executable.

PhilippKitz, 2019-10-20:
[image]

The Library Manager and the Board Manager find nothing!

moghaddam, 2019-10-20:
The same happens on my Ubuntu machine with the following stack trace:

```
root WARN EditorNavigationContribution.onStart is slow, took: 207.4000000138767 ms                                                                                                                                 
root ERROR Request repositories failed with error: Git could not be found at the expected path: '/home/ezm/applications/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/dugite-no-gpl/git/bin/git'. This
 might be a problem with how the application is packaged, so confirm this folder hasn't been removed when packaging. Error: Git could not be found at the expected path: '/home/ezm/applications/Arduino-PoC-0.0.1-
06ef598-linux/resources/app/node_modules/dugite-no-gpl/git/bin/git'. This might be a problem with how the application is packaged, so confirm this folder hasn't been removed when packaging.                      
    at /home/ezm/applications/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/dugite-no-gpl/build/lib/git-process.js:87:39                                                                              
    at exithandler (child_process.js:306:5)                                                                                                                                                                        
    at ChildProcess.errorhandler (child_process.js:318:5)                                                                                                                                                          
    at ChildProcess.emit (events.js:182:13)                                                                                                                                                                        
    at Process.ChildProcess._handle.onexit (internal/child_process.js:238:12)                                                                                                                                      
    at onErrorNT (internal/child_process.js:407:16)                                                                                                                                                                
    at process._tickCallback (internal/process/next_tick.js:63:19)
```

LinyeNinja, 2019-10-20:
The same happens to me, board and library managers don't show anything and codes output the error ""Arduino.h" not found".

cd643, 2019-10-22:
I have also this problem and the thing I found was that there isn't a folder in `Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/dugite-no-gpl/` what is named git.

rbmontes, 2019-10-23:
"An unexpected error occurred when locating the Git executable."

i fixed the problem by installing the GIT application  https://git-scm.com/download/win

cd643, 2019-10-23:
Nice, it works!

SB-68846, 2019-11-06:
Installing Git solved the issue for me to (Win10).

The error message was not very clear to me, something like "Error: Could not locate Git executable" would have pointed me in the right direction right away.

ROBERTOCARLOSALVARENGA, 2019-11-24:
"An unexpected error occurred when locating the Git executable."

i fixed the problem by installing the GIT application https://git-scm.com/download/win

kittaakos, 2020-04-08:
We [generalized the Git exec lookup](https://github.com/TypeFox/find-git-exec/issues/11) in [`find-git-exec`](https://github.com/TypeFox/find-git-exec/).

Please reopen if required.

closed 2020-04-08 by kittaakos