arduino/arduino-pro-ide #31: Installed on Ubuntu, got "not found: clangd" on launch

djsegfault, 2019-10-20:
**Describe the bug**
When I launched the program I got a "Failed to start Arduino language server: Error: not found: clangd" dialog box

**To Reproduce**
Steps to reproduce the behavior:
1 Downloaded
2 I untarred the file under /opt on my Kubuntu 18.04.3 machine
3 Ran "/opt/Arduino-PoC-0.0.1-06ef598-linux/arduino-electron"
4 Got a dialog box "Failed to start Arduino language server: Error: not found: clangd"

**Expected behavior**
No error dialog

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. 22]

**Additional context**
Here is the command line output, which includes a stack trace on that error
`david@tracer:/dload/graphics/jdownloader$  /opt/Arduino-PoC-0.0.1-06ef598-linux/arduino-electron 

(arduino-electron:19598): Gtk-WARNING **: 04:20:27.857: Theme parsing error: gtk.css:68:35: The style property GtkButton:child-displacement-x is deprecated and shouldn't be used anymore. It will be removed in a future version

(arduino-electron:19598): Gtk-WARNING **: 04:20:27.857: Theme parsing error: gtk.css:69:35: The style property GtkButton:child-displacement-y is deprecated and shouldn't be used anymore. It will be removed in a future version

(arduino-electron:19598): Gtk-WARNING **: 04:20:27.857: Theme parsing error: gtk.css:73:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version
root INFO Theia app listening on http://localhost:44067.
daemon INFO >>> Starting 'arduino-cli' daemon... [/opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/arduino-ide-extension/build/arduino-cli]
daemon INFO [INFO] Checking for config file in: /home/david/.arduino15/arduino-cli.yaml
daemon INFO [INFO] Checking if CLI is Bundled into the IDE
daemon INFO [INFO] Candidate IDE Directory: /opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/arduino-ide-extension/build
daemon INFO [INFO] CLI is not bundled into the IDE
daemon INFO [INFO] Checking for config in: /arduino-cli.yaml
daemon INFO [INFO] error loading [error: open /arduino-cli.yaml: no such file or directory]
daemon INFO [INFO] Checking for config in: /dload/arduino-cli.yaml
daemon INFO [INFO] error loading [error: open /dload/arduino-cli.yaml: no such file or directory]
daemon INFO [INFO] Checking for config in: /dload/graphics/arduino-cli.yaml
daemon INFO [INFO] error loading [error: open /dload/graphics/arduino-cli.yaml: no such file or directory]
daemon INFO [INFO] Checking for config in: /dload/graphics/jdownloader/arduino-cli.yaml
daemon INFO [INFO] error loading [error: open /dload/graphics/jdownloader/arduino-cli.yaml: no such file or directory]
daemon INFO [INFO] Configuration set
daemon INFO [INFO] arduino-cli-0.3.7-alpha.preview
daemon INFO [INFO] Starting root command preparation (`arduino`)
daemon INFO [INFO] Formatter set
daemon INFO <<< The 'arduino-cli' daemon is up an running.
root WARN please install @theia/electron@0.11.0-next.fbd63c5d as a runtime dependency
root INFO Git askpass helper is listening on http://127.0.0.1:45827.
root WARN Failed to resolve module: @theia/editor
root WARN Failed to resolve module: @theia/electron
root WARN Failed to resolve module: @typefox/monaco-editor-core
root WARN Failed to resolve module: @theia/messages
root WARN Failed to resolve module: @theia/monaco
root WARN Failed to resolve module: @theia/navigator
root WARN Failed to resolve module: @theia/preferences
root WARN Failed to resolve module: @theia/textmate-grammars
root INFO Using Git [2.17.1] from the PATH. (/usr/bin/git)
root WARN Collided keybinding is ignored;  {"command":"monaco.editor.action.toggleTabFocusMode","keybinding":"ctrl+m"}  collided with  {"command":"core.toggleMaximized","keybinding":"ctrl+m"}
root WARN Could not register keybinding:
  {"command":"monaco.editor.action.toggleTabFocusMode","keybinding":"ctrl+m"}
Error: "ctrl+m" is in collision with something else [scope:0]
root WARN Collided keybinding is ignored;  {"command":"passthrough","keybinding":"ctrl+k","context":"terminalActive"}  collided with  {"command":"terminal:clear","keybinding":"ctrlcmd+k","context":"terminalActive"}
root WARN Could not register keybinding:
  {"command":"passthrough","keybinding":"ctrl+k","context":"terminalActive"}
Error: "ctrl+k" is in collision with something else [scope:0]
root WARN Collided keybinding is ignored;  {"command":"serial-monitor:toggle","keybinding":"ctrl+shift+m"}  collided with  {"command":"problemsView:toggle","keybinding":"ctrlcmd+shift+m"}
root WARN Could not register keybinding:
  {"command":"serial-monitor:toggle","keybinding":"ctrl+shift+m"}
[... 120 lines cut ...]
    at FSReqWrap.oncomplete (fs.js:154:21)
daemon INFO [INFO] Updating libraries index
root ERROR The ws channel does not exist 25
daemon INFO [INFO] Loading hardware from: /home/david/.arduino15/packages
daemon INFO [INFO] Loading package arduino from: /home/david/.arduino15/packages/arduino/hardware
daemon INFO [INFO] Loaded platform [platform: {ID:"arduino:samd",Installed:"1.8.2",Latest:"1.8.4",Name:"Arduino SAMD Boards (32-bits ARM Cortex-M0+)"}]
daemon INFO [INFO] Checking existence of 'tools' path: /home/david/.arduino15/packages/arduino/tools
daemon INFO [INFO] Loading tools from dir: /home/david/.arduino15/packages/arduino/tools
daemon INFO [INFO] Loaded tool [tool: Version: 4.5.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
daemon INFO [INFO] Loaded tool [tool: Version: 1.2.0 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: all]]
daemon INFO [INFO] Loaded tool [tool: Version: 1.2.1 Systems: [OS: i686-linux-gnu, OS: x86_64-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: i686-mingw32]]
daemon INFO [INFO] Loaded tool [tool: Version: 7-2017q4 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu]]
daemon INFO [INFO] Loaded tool [tool: Version: 1.7.0-arduino3 Systems: [OS: i686-mingw32, OS: x86_64-apple-darwin, OS: x86_64-pc-linux-gnu, OS: i686-pc-linux-gnu, OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu]]
daemon INFO [INFO] Loaded tool [tool: Version: 0.10.0-arduino7 Systems: [OS: arm-linux-gnueabihf, OS: aarch64-linux-gnu, OS: i386-apple-darwin11, OS: x86_64-linux-gnu, OS: i686-linux-gnu, OS: i686-mingw32]]
daemon INFO [INFO] Loading package builtin from: /home/david/.arduino15/packages/builtin
daemon INFO [INFO] Checking existence of 'tools' path: /home/david/.arduino15/packages/builtin/tools
daemon INFO [INFO] Loading tools from dir: /home/david/.arduino15/packages/builtin/tools
daemon INFO [INFO] Loaded tool [tool: Version: 5.8-arduino11]
daemon INFO [INFO] Loaded tool [tool: Version: 1.0.0]
daemon INFO [INFO] Adding libraries dir [dir: /home/david/Arduino/libraries, location: sketchbook]
daemon INFO [INFO] Adding libraries dir [dir: /home/david/.arduino15/packages/arduino/hardware/samd/1.8.2/libraries, location: platform]
root INFO  <<< New client has been successfully created and cached for file:///home/david/Arduino/sketch_october20b.
discovery INFO Initialized attached boards. - Discovered boards: []
root INFO Attached boards changed:  {"oldState":{"boards":[]},"newState":{"boards":[]}}
`

gvarisco, 2019-10-20:
Hi! This can be solved by:
`snap install clangd --classic`

We will amend this requirement in the next build. /cc @mastrolinux

djsegfault, 2019-10-20:
OK I tried "sudo apt-get install clang-tools-5.0" and that didn't fix it  I just tried "sudo snap install clangd --classic" and that also didn't fix it. I got the same error

languages ERROR Error occurred while starting language contribution. /services/languages/ino/:id. Error: not found: clangd
    at getNotFoundError (/opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/which/which.js:13:12)
    at F (/opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/which/which.js:68:19)
    at E (/opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/which/which.js:80:29)
    at /opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/which/which.js:89:16
    at /opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/isexe/index.js:42:5
    at /opt/Arduino-PoC-0.0.1-06ef598-linux/resources/app/node_modules/isexe/mode.js:8:5
    at FSReqWrap.oncomplete (fs.js:154:21)

djsegfault, 2019-10-20:
$ snap info clangd
name:      clangd
summary:   Language Server Protocol implementation for C/C++ IDEs
publisher: Marco Trevisan (Treviño) (3v1n0)
license:   unset
description: |
  Clangd is an implementation of the Language Server Protocol leveraging Clang.
  Clangd’s goal is to provide language “smartness” features like code
  completion, find references, etc. for clients such as C/C++ Editors.
  
  Clangd is not meant to be used by C/C++ developers directly but rather from a
  client implementing the protocol. A client would be typically implemented in
  an IDE or an editor.
commands:
  - clangd
snap-id:      Nw2SyqKbD2SvyuOu5fFAxXJ1sjeOw8zx
tracking:     stable
refresh-date: today at 04:40 EDT
channels:
  stable:    6.0.0+git2102885  2018-01-06  (123) 6MB classic
  candidate: 6.0.0+gitbe38a81  2017-11-24   (37) 5MB classic
  beta:      6.0.0+gitbe38a81  2017-11-24   (37) 5MB classic
  edge:      10.0.0+gitc624032 2019-10-20 (2503) 9MB classic
installed:   6.0.0+git2102885              (123) 6MB classic

gvarisco, 2019-10-20:
Duplicate of #6

djsegfault, 2019-10-20:
It looks like a duplicate of #6 however in my case the suggested fix did not work

hulkco, 2019-10-25:
> Hi! This can be solved by:
> `snap install clangd --classic`
> 
> We will amend this requirement in the next build. /cc @mastrolinux

This work for me!!!

mastrolinux, 2019-11-15:
some more details, if I run just the IDE:
`root ERROR Arduino: 2019/11/15 17:38:33 Timeout exceeded while waiting for a reply from clangd: initialize` 
While if I run clangd manually everything starts to work correctly. It can be a timeout issue @kittaakos

jankeromnes, 2019-12-12:
v0.0.3 has been released, so I guess let's move this issue to the v0.0.4 milestone.

@rsora Please let me know if I can help you produce a `clangd` binary for Linux somehow.

spoenemann, 2020-01-14:
The next version v0.0.4 will bundle clangd also for Linux. Please open a new issue if there are problems running the bundled clangd with your Linux distribution.

closed 2020-01-14 by spoenemann