arduino/arduino-pro-ide #32: failed to start arduino language server: error: not found:clangd

OMGdaDPS, 2019-10-20:
Well, just as the title states. Upon downloading, then extracting, then opening the directory and running the electron app for the IDE, this error message appears after the IDE loads in. Everything seems to work fine otherwise. I thought I would just make sure that you knew this was happening (at least for me).
 I'm running KDE Neon-plasma version 4.17- (Ubuntu 18.04 based).
HP EliteBook 840 G5
i5-5300u
8GB of memory
INTEL SSDSC2BF180A4H

gvarisco, 2019-10-20:
Duplicate of #6

gvarisco, 2019-10-20:
@OMGdaDPS can you please ty `snap install clangd --classic` and let us know if that solves the issue?

OMGdaDPS, 2019-10-20:
omw

OMGdaDPS, 2019-10-20:
worked flawlessly @gvarisco . Had no idea that I had to install that via snap. I tried looking for said package via synaptic, to no avail though, before this. Thank you, and AWESOME job on the IDE. This is EXACTLY what we have been needing in the community! TYTYTYTYTYTYTYTY!

closed 2019-10-20 by OMGdaDPS