arduino/arduino-pro-ide #35: Boards Manager Bugs

Maherulazam, 2019-10-20:
Can't scroll in the Boards Manager (No Scroll Bar) and the search is not working as well. 

[image]
[image]

Can't detect the Arduino Uno connected to COM38.

[image]
[image]

rsora, 2019-10-21:
Hi @Maherulazam  , thanks for your report!

I see form your screenshot that you are on a Windows OS, could you please add the version of your Operating system to the issue description?

In the meantime, can you please check if the snapshot builds attached here https://github.com/arduino/arduino-pro-ide/issues/51#issuecomment-544560707 solve your detection problem?

Thanks!

Erriez, 2019-10-21:
Same problem for me on Windows 10 64-bit: No serial port found, but multiple serial ports are listed in the original IDE and Windows device manager.

mastrolinux, 2019-10-22:
Can you please try again with the last build available at https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.1-alpha.preview-snapshot-20191021.1 and let us know?

Erriez, 2019-10-23:
The scrollbar is now visible on my Windows 10 computer and I am able to select the Arduino UNO board with _Arduino Pro IDE Windows v0.0.1-alpha.preview-snapshot-20191021.1:_

[image]

Serial ports not detected is still a Major issue on all platforms:

- https://github.com/arduino/arduino-pro-ide/issues/75
- https://github.com/arduino/arduino-pro-ide/issues/78
- https://github.com/arduino/arduino-pro-ide/issues/86

kittaakos, 2020-04-08:
This has been fixed. I am going to close it.

----

Longer story:
Previously we collected only those boards that have an FQBN, it was incomplete logic.
Now, we collect all boards that have an FQBN, plus we collect all used ports:
```json
[
  {
    "address": "/dev/cu.SOC",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.MALS",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.Bluetooth-Incoming-Port",
    "protocol": "serial",
    "protocol_label": "Serial Port"
  },
  {
    "address": "/dev/cu.usbmodem14601",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)",
    "boards": [
      {
        "name": "Arduino MKR1000",
        "FQBN": "arduino:samd:mkr1000"
      }
    ]
  },
  {
    "address": "/dev/cu.usbserial-0001",
    "protocol": "serial",
    "protocol_label": "Serial Port (USB)"
  }
]%  
```
In the above example, there is an attached `Arduino MKR1000` ( on `/dev/cu.usbmodem14601`) and an ESP8266 (on `/dev/cu.usbserial-0001"`).

[image]

Furthermore, we have better support for 3rd party, "unknown" boards: https://github.com/arduino/arduino-pro-ide/issues/198#issuecomment-600059664

closed 2020-04-08 by kittaakos