arduino/arduino-pro-ide #36: boards and ports

jbiel09, 2019-10-20:
I can't search boards and a can't select the port. I'm from brazil, sorry i don't speak english very well but I expect help you

nikilase, 2019-10-20:
What OS are you on? If on Linux, have you added your user to the group dialout. I think you can do this with `sudo adduser YOUR_USER dialout`
Otherwise, have you installed your board in the board manager? It can be opened under Tools -> Board Manager

DrKsantor, 2019-10-21:
In windows this behavior occurs when Arduino-Pro located in Program Files (folder with administrative permissions) and run not as administrator.

rsora, 2019-10-21:
Hi @jbiel09 , thanks for your report!

Can you please check if the snapshot builds attached here https://github.com/arduino/arduino-pro-ide/issues/51#issuecomment-544560707 solve your problem?

Thanks!

pranjal991, 2019-10-22:
> In windows 10 this behavior occurs when Arduino-Pro located in Program Files (folder with administrative permissions) and run not as administrator.
> Solving ways:
> 
> 1. Run as administator
> 2. Locate IDE files somewhere else.

For me it worked in different way:
1.Run normally/administrator
2.IDE files located in C drive
 
But only boards issue is resolved ,not the ports issue still ports are not visible.
[image]

mastrolinux, 2019-10-25:
Should be fixed by https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview

mastrolinux, 2020-03-02:
Closing because no feedback

closed 2020-03-02 by mastrolinux