arduino/arduino-pro-ide #40: Can't close Board / Library Manager tab

Fede85, 2019-10-20:
Once the Board Manager or Library Manager tab is open in the IDE there's no way to close it.

**To Reproduce**
1.Click on: _Tools > Board Manager_ or _Sketch > Manage Libraries..._

**Expected behavior**
Once I'm done with installing a library or a new core I expect to close the left panel (/tab) with the the Board and Library Manager.

**Additional context**
This issue seems related to #29

nikilase, 2019-10-20:
Might be a stupid answer but have you tried clicking on the microcontroller/toolbox to close them? Microcontroller closes the Board Manager.

Fede85, 2019-10-20:
Not supid at all! I didn't noticed that by clicking on the chip icon you can toggle the Board Manager tab. Thanks for pointing out!
Maybe I was deceived by the white background same for text editor. A different color or a vertical separator for the Board Manager menu could help.

spoenemann, 2020-01-13:
It's the same approach as with VS Code: clicking on the view icon collapses it, clicking it again expands it.

More hints:
 * Press `F1` and type "Reset Workbench Layout" => resets the UI to its initial state.
 * When the Boards Manager is visible and you select Tools > Boards Manager from the menu again, the view is completely hidden (same for "Manage Libraries").

closed 2020-01-13 by spoenemann