arduino/arduino-pro-ide #42: Board Manager: can't update a pre-installed core

Fede85, 2019-10-20:
The pro IDE recognized the pre-installed cores in the "Arduino15" directory but when I open the Board Manager, shows the loaded cores as installed and I can't do more, like upgrading it.
Clicking on it has no effect.

**To Reproduce**
1. Install the "classic" Arduino IDE and install a core selecting an old version. i.e.: "SAMD 1.6.20"
2. Open the pro IDE, "Tools > Board Manager"
3. Search "SAMD"
4. No option to update it are available. 

The same apply if you want downgrade.

**Expected behavior**
A menu with the available version to chose should appear like in the "classic" IDE.

**Laptop:**
 - OS: Windows
 - Version 10

kittaakos, 2019-11-13:
@mastrolinux, this is a limitation in the CLI currently: https://github.com/arduino/arduino-cli/issues/183

kittaakos, 2019-11-21:
> this is a limitation in the CLI currently

This limitation has been resolved. For the record; we will implement this feature for the libraries as well.

kittaakos, 2019-11-22:
It is done.

kittaakos, 2019-12-11:
Was fixed in `0.0.3`.

closed 2019-12-11 by kittaakos