arduino/arduino-pro-ide #43: Arduino Pro IDE is wonderful But it is missing this feature.

OwlMatics, 2019-10-20:
Arduino Pro IDE has great features. But I noticed that it is a little hard to use.
**For Example [Support for additional languages other than C++]**
How can we use this feature? There's nothing to explain that. 
There are experienced programmers as well as new programmers. They need to understand it too.

**How You Can Solve It:**
Arduino can make a video to guide all the features.
Or
Arduino Pro IDE can have a Guide on startup which can help the newbies.

If Arduino Pro IDE is easy to use, a lot of people would love to use it.

davinderc13, 2019-10-20:
You do understand that Pro = Professional = not newbie? For newbies there is the classical IDE which is made to be simple for newbies to learn ;)

MrRSquared, 2019-10-20:
OwlMatics' point, even an advanced user (perhaps one who is not well versed in VSCode) may have difficulty navigating some of the options and features.  However, this is linked to my post a few places up.  A forum for off topic discussions about how to do x would be helpful to keep those discussions off this thread.  As this is alpha software, I would assume knowledge base, walk-throughs and the like are the last pieces to be developed.  Now the focus may be on just ensuring the software is working effectively.

OwlMatics, 2019-10-21:
> You do understand that Pro = Professional = not newbie? For newbies there is the classical IDE which is made to be simple for newbies to learn ;)

It doesn't matter who uses it. Arduino has always been easy to use. So it is expected that Arduino Pro IDE has to be simple too. ;)

dronecz, 2019-10-21:
@OwlMatics this is just alpha version of this software so it means a lot of thinks will have to be fixed. Sure it will be much more better experience when it will beta, but surely it will not be tomorrow 😉 .  
Use Arduino IDE 1.8.10 if you want things to work and do not complain here about something which does not exist yet (beta SW).

OwlMatics, 2019-10-21:
@dronecz Well maybe Arduino named it ENHANCEMENT(improvements for future release) so I thought i can help you guys to make it awesome. I mean to say, please add this feature in the stable release.

ubidefeo, 2021-02-19:
Documentation is in the works to be released when Beta comes out

closed 2021-02-19 by ubidefeo