arduino/arduino-pro-ide #44: Support saving and programming from .hex files

Siliconfarmer, 2019-10-20:
**Is your feature request related to a problem? Please describe.**
For large, complex Arduino projects that require multiple libraries to be installed, it can be difficult to remotely support installations in the field that need to be updated.  
It is also inconvenient to always need to use my desktop development system to program Arduinos, or very carefully duplicate the setup and libraries needed on my laptop so I can program multiple units in a class or other setting away from home.

**Describe the solution you'd like**
I would like to easily save the .hex file generated to a file location and name of my choosing when compiling an Arduino project.  I would like to be able to later send the file to another user, or use the file on another system of my own, and without having the source code and additional libraries installed,  be able to easily select that .hex file and program an Arduino board.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Recently, I built a complex project that was shipped across the country. As bugs were found, I updated the Arduino code and had to repeatedly ship programmed Arduinos to folks on-sight that had to partially disassemble the project to replace the Arduino board. 

**Additional context**
Add any other context or screenshots about the feature request here.

dronecz, 2019-10-20:
Hi, 

from my point of view, it would be better to have some sort of system where you can define or set some dependencies in your sketch and IDE will check if is everything ok for compilation and if not, i will install what is missing.

Example: 

I have sketch, where I use Adafruit GFX library from Library manager (LM) and some other library which is not listed in LM. When I open my sketch or if somebody else open my sketch in othe part of the world, IDE will look for these dependencies and if some library is missing, it wil ask if I want it to install or it will install automatically (this could be by checkbox in Preferences). 
I think this would work for everybody everywhere. 

I like your IDE for flashing HEX file though. I do not understand why IDE is compiling code everytime again, even when is unchanged, before it can be flashed to the board. I would welcome some option where I can export HEX file to the sketch location and when I want to flash it again, IDE will ask if can use this exported HEX file.

Siliconfarmer, 2019-10-20:
Yes, I have seen several people ask in forums online how to program from the HEX file that is always being creating somewhere in the file system during an Arduino compile. But it is hard to find that HEX file, and you need to use some system other than Arduino to then use the HEX file for programming.

It seems like a (relatively) easy feature to implement if enough people request it.

dronecz, 2019-10-20:
You do not use Project -> Export binary file (Win: Ctrl+Alt+S)? This will export binary to your sketch folder, which you can easily find by Project ->Open sketch location (Win: Ctrl+K). Sadly you have to use some other app to program your device.

per1234, 2021-02-19:
>I would like to easily save the .hex file generated

[The new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) now has a **Sketch > Export compiled Binary** feature. This will save the binary under the `build` folder of the sketch being compiled. You can quickly access that via **File > Show sketch folder**.

>be able to easily select that .hex file and program an Arduino board.

This capability is now provided by Arduino CLI's `arduino-cli upload --input-file`:
https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_upload/

closed 2021-02-19 by per1234