arduino/arduino-pro-ide #50: doesn't open on Ubuntu 19.10

OMGdaDPS, 2019-10-20:
I try to open the electron app for the IDE, after extracting, but it opens in my hex editor.

mwz86, 2019-10-20:
run it usinig terminal
sudo (location)/arduino-electron
i tested it now its working

OMGdaDPS, 2019-10-20:
omw

OMGdaDPS, 2019-10-20:
yeah, that seems to work. I wonder why I was able to just double click it and it worked, on KDE Neon, but not on vanilla Ubuntu 19.10

mwz86, 2019-10-20:
now i'm testing it 
there is some problems:
Failed to start arduino language package
Error: no core is installed for arduino uno #_#

OMGdaDPS, 2019-10-20:
i was told to install clangd by using sudo snap install clangd --classic. it worked for me

OMGdaDPS, 2019-10-21:
What did you do, just untar the archive and put it somewhere, go into the folder and double click the file labeled "arduino-electron"?

OMGdaDPS, 2019-10-21:
I think it has something to do with maybe me installing a few other programs, and that has done something. Because like I said, it tries to open in my hex editor. I of course can choose a different program to try and open it with, but none have worked so far. The closest that I have gotten to it working was using "run software" in the "open in other application". It gives me a spinning "working in background circle, but the doesnt do anything.
   This is all of course, only on one of my machines. I have tested it on one other laptop so far, and it worked without a hitch on that one. I did exactly what I asked you if you had done.

kittaakos, 2020-04-08:
> there is some problems:
> Failed to start arduino language package

> i was told to install clangd by using sudo snap install clangd --classic. it worked for me

I can confirm this with older versions. Now, we ship the app with `clangd` bundled into it.

> What did you do, just untar the archive and put it somewhere, go into the folder and double click the file labeled "arduino-electron"?

See [this thread](https://github.com/arduino/arduino-pro-ide/issues/160#issuecomment-561216286), @OMGdaDPS.

Please reopen if required.

closed 2020-04-08 by kittaakos