arduino/arduino-pro-ide #51: Doesn't detect some attached boards

OMGdaDPS, 2019-10-20:
When attaching a board to the PC, while the IDE is running, I have tried 14 different Arduino Nano's, 2 Arduino M0's, 1 Arduino Mega 2560, CPX from adafruit. 
  The only one that doesn't show up in the "boards" dropdown (right next to 'save') is the Nano.

rsora, 2019-10-21:
Hi @OMGdaDPS, thanks for your report!

Can you please check if the ubuntu snapshot build attached here https://github.com/arduino/arduino-pro-ide/issues/54 solves your problem?

Thanks!

OMGdaDPS, 2019-10-22:
Just tried the new snapshot, and it didn't fix it for me. For some reason it is still not detecting my Nano boards. It is however detecting my other boards still. I also have a new error showing up about not being able to find the git executable or something. ill open another bug

mastrolinux, 2019-10-28:
https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview should fix it.

OMGdaDPS, 2019-10-28:
give me just a little bit, im walking to my workshop now. then ill warm everything up and give it a go. i know that may sound weird, but i have a lot of hardware setup in there (so much so that i literally use my 6 car garage lmao), and i guess this would be the perfect thing to test this build out on. 
  I also am free ANYTIME to do other tests as well, on whatever arch (x86, arm, etc) that people may need. just hmu.
  omw

OMGdaDPS, 2019-10-28:
ok so it seems to be detecting all of my boards now! yay! thanks!

closed 2019-10-28 by OMGdaDPS