arduino/arduino-pro-ide #52: THIS IS WHAT WE NEED.

IMSHOX, 2019-10-20:
It’s good, but we do not need IDE like VS CODE and we already have it (VScode , Atom ……), I’m an electronics engineer, we need a HARDWARE debugger not an interface for computer science.

we need to see A BOARD with I / O declarations (INPUT / INPUT_PULLUP, OUTPUT), see the level of memory usage and what if we do new bootloader that enable hardware debuging (assembler to C ++ debuging) I know that the execution will be very slow but the debug will exist for the first time.

a strong IDE for the Arduino requires a lot of connection between the HARDWARE and the SOFT.

IMSHOX, 2019-10-20:
If we have to design, I will design the interface.

davinderc13, 2019-10-20:
The product has been developed, maybe we should focus on the intended uses by the developers and not create a new product?

IMSHOX, 2019-10-20:
I didn't say to change everything, just to go in the right direction,
We have the base, so we have to load it with good things.

davinderc13, 2019-10-20:
This is the alpha release = needs lots of feedback to get a stable operational base. This isn't the place to request major features. But hey, if you'd like to develop something new, it seems to code is there, just branch it and create the feature you need. Maybe they'll even merge it if you do a good job ;)

nikilase, 2019-10-20:
I think they have planned on adding a debugger to the IDE. At least they say so on their blog entry

Catman155, 2019-10-21:
The “intended uses” don’t seem to add anything useful. The fact this is based on electron makes me sick. Who wants 100 chromium instances running on their pc taking up 16GB RAM. As long as the features don’t compensate for the technical dept. it isn’t worth my time. 

Board overview and being able to see the status on all pins. Something like that. That would be nice. That would make it not useless.

per1234, 2021-02-19:
A debugger has been added.

closed 2021-02-19 by per1234