arduino/arduino-pro-ide #55: Not a suggestion or a bug, just a question.

OwlMatics, 2019-10-21:
When will Arduino Pro IDE support multiple languages other than C++?
Will Python be one of them?

Everyone loves Arduino because of its easy to use features but if it supports programming languages like Python, people would be mad after it.

mbanzi, 2019-10-21:
" people would be mad after it." not sure what you mean there

OwlMatics, 2019-10-21:
I mean they would love it.

torenrl, 2019-10-22:
I guess Python might be supported at some point, considering the work Adafruit has done with MicroPython, and the available boards supporting this from f.ex. Adafruit and SparkFun. However, I can imagine this is not the priority at this point since Arduino does not have similar boards (yet?). The multi-language support in Arduino Pro IDE however hints towards future support for this, and the possibility to extend the IDE is probably going to be looked into for this purpose by, if not others, at least Adafruit engineers. I would not worry about this, with some patience, you'll get what you wish for...

Even better, how about looking into this yourself, and contribute to the development once the IDE is open-sourced?

OwlMatics, 2019-10-22:
@torenrl I didn’t get the last paragraph. How can I contribute myself?

torenrl, 2019-10-22:
When the IDE is open-sourced, and we know how the IDE extensions are made, you could (together with others of course) develop an extension to the IDE which includes and calls the appropriate CLI commands to upload to a MicroPython board.

I agree by the way that MicroPython should be available through the Arduino IDE at this point, considering the popularity of the language.