arduino/arduino-pro-ide #58: It should possible to create user color themes like in VSC and to import VSC themes

sstaub, 2019-10-21:
**Describe the solution you'd like**
On VSC you have the command "Generate Color Theme from Current Setting" at a starting point 
to make own color themes. Maybe this is also possible in Theia.
It should also possible to import themes from VSC or Sublime Text

**Additional context**
[image]

svenefftinge, 2019-11-28:
Info: VS Code Theme support will land in Theia in December.

sstaub, 2020-09-14:
Any news about?

ubidefeo, 2021-01-15:
@AlissonBonatto 
no news on our end
for now our priorities are geared toward functionalities and bug-fixing
hope you'll understand :)

AlissonBonatto, 2021-01-15:
@ubidefeo 

No problems! Thanks for the feedback :)

ubidefeo, 2021-02-19:
Issue linked to #199 which will be closed in favour of this