arduino/arduino-pro-ide #61: ESP32 Dev module can't detect

nomancyclewala, 2019-10-21:
the normal Arduino ide easily detect ESP32 dev module but 
the new pro ide can't detecte it and say "no esp32 board is detected "

[image]

sirapol, 2019-10-28:
I can't upload to ESP32.

It's report like a picture.

[image]

hanif85, 2020-06-21:
[image]
just open arduino-cli.yaml and add [https://dl.espressif.com/dl/package_esp32_index.json](url)

RoyHopkins, 2020-08-28:
Please explain how to install the ESP32in the Arduino Pro

ubidefeo, 2020-08-28:
@RoyHopkins 
the comment just above yours tells you how
you need to add the index url to the configuration file and you'll be able to find the board in the Boards Manager

kittaakos, 2020-09-03:
> Please explain how to install the ESP32in the Arduino Pro

See here: https://github.com/arduino/arduino-pro-ide/issues/245

I am going to close this issue, let me know if you want me to reopen but you can also create a new one.

closed 2020-09-03 by kittaakos

enavascalmell, 2020-09-22:
Hi, How do you install a Board esp32 on Arduino pro IDE?

per1234, 2020-09-22:
@enavascalmell This issue tracker is only to be used to report bugs or feature requests. This is not the place to ask for support using the Pro IDE. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules.