arduino/arduino-pro-ide #62: Remove default Electron options from topbar

DoodlesEpic, 2019-10-21:
The top bar still has the default options from Electron like (cut, copy, paste, etc...), those are extremely unnecessary and only add to the clutter in the design. Since it's a pro application everyone using should know the shortcuts to copy and paste.

spoenemann, 2020-01-13:
What do you mean by "top bar"? The _Edit_ menu?

DoodlesEpic, 2020-09-04:
I was originally referring to all options in the _File, Edit, and Help_ menu, I guess it comes down more to preference, and as the IDE still has lots to improve (such as actually releasing the source code in this repository), I think it's not a top priority to consider this issue.

closed 2020-09-04 by DoodlesEpic

ubidefeo, 2020-09-05:
@DoodlesEpic 
the menu has been completely overhauled as of `0.1.0` and is now more aligned with our legacy IDE