arduino/arduino-pro-ide #63: Sketchbook missing

PaulStoffregen, 2019-10-21:
**Describe the bug**
The sketchbook list does not appear in the File menu.

**To Reproduce**
Use Arduino 1.8.10 and save some sketches.

Then run this new IDE and click the File menu.  No sketchbook list.

**Expected behavior**
I was expecting the new IDE to find my sketchbook location from the old IDE and show it in a similar way.

**Desktop (please complete the following information):**
Linux, Ubuntu 18.04, 64 bit

PaulStoffregen, 2019-10-21:
Also looks like saving a few defaults to a location, rather than where I had configured my sketchbook using File > Preferences in the old IDE.

mastrolinux, 2019-10-27:
The sketchbook is visible in the open file icon.

kittaakos, 2021-02-11:
It's under `File` > `Sketchbook`.

closed 2021-02-11 by kittaakos