arduino/arduino-pro-ide #64: Make it open-source

DoodlesEpic, 2019-10-21:
Let us contribute to the project, as-is development look slow, allowing for pull requests via Github or any other way to contribute would make the project develop much faster with a healthier community.

joelvonrotz, 2019-10-21:
[image]

Consider reading the blog post. They want to create the basics of the tool before releasing the source code.

DoodlesEpic, 2019-10-21:
I've read it, I just wish the alpha version was also open-source.

joelvonrotz, 2019-10-21:
> 
> 
> I've read it, I just wish the alpha version was also open-source.

oh alright, I apologize. Well I think they want to complete the alpha, as in include all the features they have planned. After the alpha and the release of the software, I'd imagine optimizations could be suggested.

DoodlesEpic, 2019-10-21:
Sure, that does make sense, this topic has nothing to do so I'll just close it.

closed 2019-10-21 by DoodlesEpic