arduino/arduino-pro-ide #65: Intellisense problems

DoodlesEpic, 2019-10-21:
The following lines generate intellisense errors:
* pinMode(0, OUTPUT) -> Use of undeclared identifier 'OUTPUT'
* byte pinNumber -> Unknown type name 'byte'

It looks like a few of the Arduino processing language keywords are not yet working correctly with the editor.

spoenemann, 2020-01-13:
These are not language keywords, but symbols defined in C/C++ header files (`.h`)

What board have you selected?

Blubbaa, 2020-09-01:
Using functions from included namespaces, i.e. `Communication::init(callbackOn, callbackOff);` also results in an error message in the main sketch: `Use of undeclared identifier 'Communication'clang(undeclared_var_use)`. Compiling and uploading works fine.

Selected board is ESP32 Pico Kit.

DoodlesEpic, 2020-09-04:
Hey, @Blubbaa this is a VERY old issue. I think the original issue I posted here has been fixed, one would think this issue would be automatically closed by a bot, but they have not added one to the repository yet, so I'm going to close it myself.

closed 2020-09-04 by DoodlesEpic

RussellAult, 2020-10-15:
I'd like to re-open this bug; I'm having exactly the same issue with 0.1.2 on Windows 10. Create a new sketch, set the board as "Arduino Uno", and type `byte test;` into the first line and it throws a "Unknown type name 'byte' clang(unknown_typename)".

ubidefeo, 2020-10-17:
@RussellAult 
we're in the process of overhauling our Language Server module and implementing a more solid support for code hinting/completion/assist.
Next release should behave much better, no ETA yet, but it's not far :)