arduino/arduino-pro-ide #67: Linux Installation and run instructions please

FlailAway, 2019-10-21:
**Is your feature request related to a problem? Please describe.**
Some Linux Installation instructions would be nice.

Mint 19+Cinnamon.
I have downloaded the Linux version and unpacked it, but now cannot find how to start anything. I am not a newbie to the Classic Arduino IDE, but am a little lost here with Mint.

Do I need to install Eclipse etc or what. Pretty sure you will get more comments if more people can easily install it to try.

Nikurasuu, 2019-10-21:
I don't know if you found out how to open it, i only did it in ubuntu 18.04.3.
It's pretty easy you only need to open the terminal in the folder of the Arduino Pro IDE and run "./sudo arduino-electron" then it should open!

Cheers!

FlailAway, 2019-10-21:
Ahhh, OK, thanks but that crashes with " failed to find language server: Error: not found: clangd" so looks like it needs an Internet connection. Our home network PC is not online so I'll have to find how to load clangd offline.

OK, brought the PC inside and 

sudo apt update
sudo apt install clang-tools-8
sudo apt upgrade
reboot

but still the same error. Seems too raw an Alpha to be bothered with at this point.

gdsports, 2019-10-21:
Does not need to run as root at least on my Ubuntu 18.04 system.

Solution to clangd problem: https://github.com/arduino/arduino-pro-ide/issues/6

kittaakos, 2020-04-09:
> It's pretty easy you only need to open the terminal in the folder of the Arduino Pro IDE and run "./sudo arduino-electron" then it should open!
> 
> Cheers!

👍 Here are the steps once more: https://github.com/arduino/arduino-pro-ide/issues/160#issuecomment-561216286

> Ahhh, OK, thanks but that crashes with " failed to find language server: Error: not found: clangd"

We ship `clangd` with the Pro IDE for all supported platforms.

Please reopen if required.

closed 2020-04-09 by kittaakos