arduino/arduino-pro-ide #68: Board manager and library manager don't work on Windows 10 18362.418

Nikurasuu, 2019-10-21:
I can't use the board manager on **Windows 10 18362.418**, when I open the board manager **nothing shows up** (the same for the library manager).
The funny thing is when i use **Ubuntu 18.04.3** everything shows up, I can install boards and librarys and compile the sketch. 

[image]

juanesponja2, 2019-10-22:
same problem

benibalogh, 2019-10-24:
Same here, empty Boards Manager and Library Manager, plus no ports show up in "Select Other Board & Port".

RobDeVoer, 2019-10-24:
Not happening in all versions of Windows 10. Boards and Library manager work fine on both of my Windows 10 systems. Did you ensure that the required port in the Windows Firewall was opened so that the application can access the internet to load the boards/libraries?

juanesponja2, 2019-10-25:
in my case i'm sure that the Arduino-PoC.exe has internet connection

Nikurasuu, 2019-10-25:
Update:
Since the new update to version 0.0.2 my bug has been fixed and everything works perfect now! 👍

closed 2019-10-25 by Nikurasuu