arduino/arduino-pro-ide #69: Arduino UNO board not available after installation "Arduino AVR boards"

Erriez, 2019-10-21:
The Arduino UNO board is not available after "Arduino AVR boards" installation on Windows 10 64-bit.

**To Reproduce**
Steps to reproduce the behavior:
1. Start new Arduino IDE
2. Click Tools | Board manager...
3. Install "Arduino AVR boards by Arduino" (Installation successful)
4. Select other boards & ports | UNO board grayed out. Restart IDE does not help.

**Expected behavior**
Arduino/Genuine board must be available (grayed out) including a serial port (nothing found).

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 64-bit

rei-vilo, 2019-10-25:
At launch, the IDE displays a cryptic message.

[image]

Might be related.

Arduino Pro IDE Alpha 0.0.2 on macOS 10.15.

mastrolinux, 2019-10-28:
@rei-vilo no, that is not related and the error is coming from the language-server. Are you still unable to view the serial ports with https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview version? @Erriez

rei-vilo, 2019-10-28:
The port shows now on with Alpha 0.0.2. I've close #20.

Erriez, 2019-10-28:
Serial port is now visible in board window. It is possible to upload a sketch, but the serial terminal does not work. The serial port is not opened and the output window remains empty.

[image]

The sketch below works perfect with the original Arduino IDE:
```
void setup() 
{
    Serial.begin(115200);
    Serial.println(F("Hello world!"));
}

void loop() 
{
    Serial.println(F("."));
    delay(1000);
}
```
[image]

rei-vilo, 2019-10-28:
The Arduino Nano 33 BLE does print on the console under macOS when using Alpha 0.0.2.

[image]

Erriez, 2019-10-28:
I'm on Windows. The serial port is not opened for some reason.

kittaakos, 2020-04-09:
There are multiple things logged to one GH issue. Let me clarify them:

> Arduino/Genuine board must be available (grayed out) including a serial port (nothing found).

Now it is.

[image]

----

> At launch, the IDE displays a cryptic message.
> 
> [image]
> 
> Might be related.

No, it's unrelated.

----

> The serial port is not opened and the output window remains empty, no matter which serial baudrate is selected.

The monitor had some issues when the line was not terminated with `\n`. It was fixed via #201.

closed 2020-04-09 by kittaakos