arduino/arduino-pro-ide #70: Examples menu disappeared on Windows 10

Erriez, 2019-10-21:
The examples menu is not available in the File menu on Windows.

Original IDE:
[image]

New IDE:
[image]

What is the reason for this?

tigoe, 2019-11-06:
+1 to this too. I use the File menu extensively in the original Arduino app, specifically the examples menu. Removing it is a step down from the previous app.

minzukas, 2020-01-10:
following

JeremySCook, 2020-03-18:
Would request this as well. 

Love where you're going with the IDE - should be awesome when done!

Erriez, 2020-03-28:
The missing examples menu is one of the core features to quickly start a new project. Any news on this?

Erriez, 2020-09-10:
Almost a year later after the first "pro" release, we still miss `Examples` and `History` from the `File` menu. When I create libraries for Arduino, I always spend a lot of time to include examples to demonstrate how the library / hardware works, but they are not visible.

Opening examples via the `File | Open` menu is really a pain, even in the latest version 0.1.1:

Let me explain our user experiences:

1. Opening examples looks like only possible via the `File | Open` menu and browse to a `.ino` file. This is time consuming, because double clicking on a `.ino` file in the `Nautilus` file manager does not work.
2. Bug: The serial port must be closed manually before opening the next `.ino` file, otherwise the serial port is still in use.
3. Bug: After opening the new `.ino` it displays: "Error: No boards selected. Please select a board.`. The IDE should remember the previous selected board.

Repeat these steps over and over again to switch between examples.

Can you please increase priority to implement a simple examples and history menu?

kittaakos, 2020-09-10:
> Can you please increase priority to implement a simple examples

We're going to roll out a new release with the built-in examples in a few days. On top of that, we will also support library examples and the include library feature. Thanks for your patience and stay tuned.

kittaakos, 2021-02-11:
It's available.

closed 2021-02-11 by kittaakos