arduino/arduino-pro-ide #72: Sketch export with all libraries (also board versions)

hutje, 2019-10-21:
**Is your feature request related to a problem? Please describe.**
When I work with different sketches on different computers, I want to be able to save the while project, with all the essential data to be able to compile the sketch on another computer.

**Describe the solution you'd like**
A easy way to save all included libraries and board with the sketch. Like a project. To be able to open and compile it on another computer.
There already is an export function, but this only includes the ino file, but this needs to be extended.

Master811129, 2019-10-27:
Yeah thats a good idea. I'm waiting for this IDE just for this feature:

"save all included libraries and board with the sketch. LIKE A PROJECT."

hutje, 2020-02-04:
Really hope this feature will be implemented... Most libraries rely on specific other library versions, this can cause problems with different projects and different library versions... Atom and Platform.io do have this option, but I really like to use the original IDE as it is intended for Arduino boards.
It just misses this one (important) feature.

per1234, 2021-02-18:
Related: https://github.com/arduino/arduino-cli/issues/108