arduino/arduino-pro-ide #75: Board/Port not being detected

notkevinjohn, 2019-10-21:
**Describe the bug**
I have an Arduino Mega ADK connected to the computer. This board is successfully detected by the classic version of the Arduino IDE. The device is location at /dev/ttyACM0 on my computer. When I attempt to select the board, no port is detected. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Board Manager
2. Install drivers for Arduino Mega AVR Boards
3. From board selection pull down, select Other Board & Port
4. Select Arduino Mega AVR
5. See Error

**Expected behavior**
The system would have allowed me to select the port /dev/ttyACM0

[image]

**Desktop (please complete the following information):**
 - Ubuntu Mate
 -  19.04

rluberto, 2019-10-21:
The latest snapshot worked for me today (located below), however I am running Windows 10. 

[https://github.com/arduino/arduino-pro-ide/issues/54#issuecomment-544558918]

rsora, 2019-10-22:
Hi @notkevinjohn, thanks for you bug report!

We just released the [build snapshot 20191021.1](https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.1-alpha.preview-snapshot-20191021.1)  in the release page, that includes board detection bugfixing. 

Can you try the new release and check if your issue is solved?

notkevinjohn, 2019-10-22:
@rluberto rsora

Confirmed that this snapshot solves the issue I documented.

mastrolinux, 2019-10-25:
Should be fixed by https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview in even a better way.

closed 2019-10-25 by mastrolinux