arduino/arduino-pro-ide #78: Problems with the label showing port connectivity

torenrl, 2019-10-22:
**Describe the bug**
The port keeps becoming labelled as disconnected even though the connection seems to be in order.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect to a board/port
2. The board/port is labelled as connected and uploading sketches and serial port monitoring are both possible
3. The port is soon labeled as disconnected, even though the I am still able to upload sketches and monitor the serial port.

**Expected behaviour**
That the port is labelled as connected

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS catalina
 - Version https://github.com/arduino/arduino-pro-ide/issues/54#issuecomment-544558918

eclipse1985, 2019-10-22:
Same on `Mojave`

mastrolinux, 2019-10-28:
Can you please try the new https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview version?

torenrl, 2019-10-29:
Seems to be working now

closed 2019-10-29 by torenrl