arduino/arduino-pro-ide #82: Library Manager: scroll does not reset on search that returns less items

FrankGrimm, 2019-10-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on "Sketch"
2. Click on 'Library Manager'
3. Scroll down using the mouse wheel.
4. In the search field enter a specific library name that only returns one result, e.g. "fastled".
5. The search result is not visible until you manually scroll up.

**Expected behavior**
The scroll state of the container should reset appropriately to show search results.

**Screenshots**
-

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 18

**Additional context**
- It might also be a general bug with scrolling in that component, with larger search results I do not see a scroll bar on the right hand side while it is there in the bugged state.

kittaakos, 2019-11-13:
Thank you for reporting the issue, @FrankGrimm. The fix is on the way and will be available with the next release.

[image]

kittaakos, 2019-12-11:
Was fixed in `0.0.3`.

closed 2019-12-11 by kittaakos