arduino/arduino-pro-ide #83: Startup hang with "SyntaxError: Unexpected token p in JSON at position 0"

todbot, 2019-10-22:
**Describe the bug**
Double-clicking on "/Applications/Arduino-PoC" on MacOS 10.14.6 results in spinner in a startup window, forever.  If run from the command-line in Terminal, the [JSON syntax error is visible as in this gist](https://gist.github.com/todbot/b7510a2acc334d946d6f2965f0d52821).
This occurs on both releases as of 22 Oct 2019.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '/Applications' in Finder
2. Double-click on 'Arduino-PoC' app
3. See spinner window appear and never go away

Or.
1. In Terminal, "cd /Applications/Arduino-PoC.app/Contents/MacOS/"
2. In Terminal, execute binary "./Arduino-PoC"
3. See error messages in Terminal, as in this gist: https://gist.github.com/todbot/b7510a2acc334d946d6f2965f0d52821

**Desktop (please complete the following information):**
 - OS: MacOS 10.14.6
 - Version v0.0.1-alpha.preview-snapshot-20191021.1 and v0.0.1

**Additional context**
I have a rather extensive Arduino set of libraries and boards, but I thought nothing too weird.

todbot, 2019-10-22:
I think I found a cause of this bug.
Previously-installed old versions of `arduino-cli` will cause `Arduino-PoC` to hang. 
I had `arduino-cli` v0.3.7. I updated it to v0.5.0 and `Arduino-PoC` starts up.

mastrolinux, 2019-10-23:
Correct, the IDE uses the System CLI if in the PATH, otherwise uses its own one. Please always use the latest release of the CLI.

closed 2019-10-23 by mastrolinux