arduino/arduino-pro-ide #84: In normal mode: can not add folder to workspace

cd643, 2019-10-22:
When you are in the normal mode, I want to add folder to workspace but nothing happens.

1. go to the explorer in arduino pro
2. Click on add folder to workspace (more actions -> add folder to workspace)
3. select a folder

after this, there was no folder in my workspace but if I drag it to the workspace it works!

 - using linux xubuntu 18.04 LTS

kittaakos, 2019-11-12:
> When you are in the normal mode, I want to add folder to workspace but nothing happens.

This behavior is expected. The normal mode has limitations; the workspace cannot be changed. We will disable this command to avoid confusion.

jankeromnes, 2019-11-12:
I can take a look at this. @mastrolinux could you please assign me to this issue?

jankeromnes, 2019-11-12:
I was able to reproduce this issue on Mac OS.

I believe the root problem is that in "non-advanced" mode, the File Explorer should not be visible at all.

On Mac at least, the Explorer is not visible by default in "non-advanced" mode. However, I found a "hack" to make the Explorer show up, by right-clicking on a editor tab handle, and clicking "Reveal":

[image]

[image]

I think that we should:
- Remove the "Reveal in Explorer" menu option in "non-advanced" mode
- Make sure that the Explorer can't show up in "non-advanced" mode

On the other hand, in "advanced" mode, "More Actions > Add Folder to Workspace" seems to work fine:

[image]

kittaakos, 2019-11-21:
> 2\. Click on add folder to workspace (more actions -> add folder to workspace)

For the record:
 - `Add Folder to Workspace...` is a different thing.
 - We will add a menu item to `File` > `Create Folder` which will open the `Explorer` in classic mode and creates a new folder inside the current sketch dir.

kittaakos, 2019-11-21:
The fix will be available in the next release.

[image]

closed 2019-12-09 by kittaakos