arduino/arduino-pro-ide #86: Does not see the Board or com port. Windows 10

tekasijimo, 2019-10-22:
Hello everyone! I have a problem with the new IDE. When choosing my Board I have no way to select the port to which the Board is connected. Installed environment on Windows 10 version 1903. 
A similar situation was already with a person with Mac OS but he was helped by reinstalling, this option unfortunately did not help me...
At the same time in the system in the device Manager the Board is, as well as the classic arduino ide sees the Board.

sstaub, 2019-10-22:
Did you try the latest snapshot release?

mjs513, 2019-10-23:
Downloade 0.0.1 - which I think is the latest and have the same problem.  Windows 10 does not see a Uno attached to the PC and no boards available for download.

I take it - it will only download a board core if that board is attached and its not already installed.

fcebreros, 2019-10-23:
Hi,
I'm also on Windows 10 (build 18362) and had success with the latest version.
Could install the board and upload a basic sketch to an Arduino Uno.

mjs513, 2019-10-23:
Yep - got that error as well.  I am on a Windows 10 x64 Home, build 17763.805 version 1809.  Maybe that will help.

tekasijimo, 2019-10-23:
I plugged in the arduino uno and ran the  latest snapshot release but I can't choose the port.
[image]

HuubBuis, 2019-10-23:
Have the same issue, no way to select a port. Running on Windows 10 release 1903, all updates installed. Tested using arduino uno (CH340) and arduino leonardo. No errors in the list of loaded modules.

arduino-electron 0.0.1

mjs513, 2019-10-23:
Since I am the only one that seems to have the problem of not seeing any boards as well as the com port issue - is there something I have to do on the install side.  All I did was download the zip and copy and paste the IDE to its own directory.

fcebreros, 2019-10-23:
> 
> 
> Since I am the only one that seems to have the problem of not seeing any boards as well as the com port issue - is there something I have to do on the install side. All I did was download the zip and copy and paste the IDE to its own directory.

It's mostly taken granted but d you need an Internet connection, you have perhaps a firewall blocking the Ide to get online?

mjs513, 2019-10-23:
Yeah - figured I would need a internet connection - that's how the ide updates the boards and libraries. :)

Just check Norton 360 and its not blocking the IDE.  Even set up a rule to specifically allow but still no luck.  Oh - can't find my libraries either.  

Maybe its the download?

benibalogh, 2019-10-24:
Same issue here, Boards Manager does not work and cannot select port on Win 10 Pro 1909 build 18363.449 with latest IDE (v0.0.1-alpha.preview-snapshot-20191021.1). Disabled firewall but nothing changed.

mjs513, 2019-10-24:
Did as @benibalogh suggested - no luck.

Could this be a Git problem - I have GIT Desktop installed but its saying GIT executable not found?

fcebreros, 2019-10-24:
Strange, now it does not work for me any more.
Can't see boards or ports but was able to see them.
No settings have changed in my machine (don't know if
Windows has done something)

tekasijimo, 2019-10-24:
In General, I had a free minute, and I tried several options. in the beginning, it helped me to run the arduino cli daemon from the command line manually, and then run the arduino electron. After that there were both boards and ports of all my connected boards. after that I copied the arduino electron to the root of the disk "c:\arduino-pro-ide" and there same copied arduino cli (broke a that Arduino-PoC.exe and arduino-cli.exe are in the same folder), and assigned Arduino-PoC.exe the rights of administrator. After that, everything worked immediately and without manipulation of the command line. I think that in archive not embedded (or path to him are worth wrong) arduino cli. 

[image]

But I still don't see my boards on the CH340G...

HuubBuis, 2019-10-24:
Tried above solution didn't work but i found an error regarding git.

> root ERROR Request repositories failed with error: Git could not be found at the expected path: 'D:\ArduinoPro\resources\app\node_modules\dugite-no-gpl\git\cmd\git.exe'.

This folder does not exist: 

> D:\ArduinoPro\resources\app\node_modules\dugite-no-gpl\git

fcebreros, 2019-10-25:
All commented suggestions didn't work for me neither.
But I had some strange behaviour. 
Because the ide stopped working I erased the workspace and settings I had saved in the default
Documents/Arduino folder. I tried opening the ide again (that I had  pasted in a folder named ArduinoPro in the ProgramFiles folder) and it didn't load. When I tried again with the folders unpacked in the original Downloads folder it worked, even the first version worked. I tested compiling and uploading without problems.
(still have the Git Executable error)

mjs513, 2019-10-25:
Just installed latest (v0.0.2-alpha.preview) on my windows 10 x64 and it worked like a charm.  Got all the boards listed as well as the libraries populated.  Selected UNO and it did the install the Core no problem.

Still have the issue with git executable mentioned by others earlier.

I just have one question - where does it install the core files.  Before they were easy to find - wanted to see the directory structure.

fcebreros, 2019-10-25:
Board is recognized.
Compile is OK.
Git executable error also still there.
Unable to upload code to board.
It get's stuck.
[image]

mastrolinux, 2019-10-25:
I will close this one, there is https://github.com/arduino/arduino-pro-ide/issues/99 about git. About the upload please open a new issue.

closed 2019-10-25 by mastrolinux