arduino/arduino-pro-ide #87: Library path

fcebreros, 2019-10-23:
I'm trying to add my own library path.
I have managed to add the folder to the workspace
but on compilation I get errors because the Ide is not
seeing the libraries.
Is there some additional configurations
that must be done?
Latest rlease: v0.0.1-alpha.preview-snapshot-20191021.1
Windows 10.0.18362

MicheleCioccarelli, 2020-03-02:
Have you tried including the library using an absolute path in ""?
ex: #include "directory/file.h"

can you show the code?

ubidefeo, 2020-03-03:
@fcebreros 
your best option for now is to add your libraries to the sketchbook `libraries` folder, unless one of my colleagues can confirm that we can already support multiple library paths
@kittaakos @cmaglie

OmarAlamoudi, 2020-06-04:
How to change the default "Sketchbook location"? Right now I have my Arduino IDE read from another directory, not the default "~/Documents/Arduino/libraries", but I don't how to change that in the Arduino Pro IDE 

I'm using Mac OS. Thanks

per1234, 2021-02-18:
[The new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) now allows you to configure the sketchbook folder location via the **File > Preferences > Sketchbook location** preference, consistent with how it's done in the classic Arduino IDE.

closed 2021-02-18 by per1234