arduino/arduino-pro-ide #90: Sometimes the IDE doesn't start

sstaub, 2019-10-23:
**Describe the bug**
The start of the IDE is very slow, sometimes never starts (see screenshot).
Sometimes you can see that the IDE UI is loaded for a short time, than the white screen comes
up, after a while (~6s on my MacBook Pro) the UI is ready but sometimes not.

**Expected behavior**
The startup time should much faster

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - macOS Catalina
 - Version 0.0.1.20191021.1

mastrolinux, 2019-10-23:
Can you also tell us if you have already an arduino-cli in your PATH? It could conflict if you have a version < 0.5.0. Eventually just remove your arduino-cli from the PATH.

type `which arduino-cli` in a terminal to verify if you have it in the PATH and `arduino-cli version` to check the version you have installed.

sstaub, 2019-10-23:
In the terminal I get no response, wrong commands?
[image]

mastrolinux, 2019-10-27:
Please try again the v0.0.2 and let me know

sstaub, 2019-10-27:
With 0.0.2 the IDE starts every time, the start up time for the GUI is now about 2s with a MacBook Pro 15'' 2018.

mastrolinux, 2019-10-28:
So I will close this issue because now it is reliable. We will work later on performance but it is not something we will fix in the alpha phase. Thank you for the feedback.

closed 2019-10-28 by mastrolinux