arduino/arduino-pro-ide #91: Error: No core is installed for Arduino Uno.

chovanj, 2019-10-23:
**Describe the bug**
After installing Arduino AVR Boards and selecting Arduino UNO from the list, it's still shouting **"Error: No core is installed for Arduino Uno. Please install the board"**.

Steps to reproduce the behavior:
1. Open a new sketch for the first time.
2. Install Arduino AVR Boards.
3. Select Arduino UNO.
4. Click on Verify.

[image]

[image]

 - OS: Windows 10 Pro
 - Version: 1903
- Arduino Pro IDE: v0.0.1-alpha.preview-snapshot-20191021.1

mastrolinux, 2019-10-23:
We are not able to reproduce, are you able to provide the full output of the developer tool? 
Ctrl + Alt + I in the app to open the Developer Tools.

chovanj, 2019-10-23:
It's a very strange because even I can't do :-). 

When I try again to select the board "Arduino Uno" disappeared and only "Arduino/Genuino UNO" stays which works after selecting. No idea how "Arduino UNO" appeared in the list for the first time and why it disappeared.

Fede85, 2019-10-23:
Same issue here despite it's a MKRZero:
[image]

mastrolinux, 2019-10-23:
@Fede85 is it just the pop up or the sketch does not compile?

Fede85, 2019-10-23:
The pop-up appear when I click the verify button. And the sketch doesn't compile.
The SAMD core was installed from the ide 1.8.x some time ago (it isn't the latest version). But should be ok for the MKRZero.
Maybe if you want to try to replicate there, you can install the same version of SAMD core  through the Board Manager of the "normal" IDE and then try to open the PRO IDE.

BTW, I'm using Ubuntu 19.10

RobDeVoer, 2019-10-24:
I also had the Arduino Uno popup appear. It disappeared after installing Arduino AVR Boards. Could not compile until after I closed ide and restarted.

(Using Windows 10)

mastrolinux, 2019-10-25:
Please try https://github.com/arduino/arduino-pro-ide/releases/tag/v0.0.2-alpha.preview and let us know

Fede85, 2019-10-29:
[image]

Still the same.

It may help: I had the same behavior with the **Arduino nrf528x** core. After updating to the latest with the classic Arduino IDE I was able to compile with the pro-IDE

greggwardnz, 2019-10-30:
Same for Nano. This is on Ubuntu 19.10 using v0.0.2 of Pro IDE:

[image]

ubidefeo, 2020-08-26:
@abionics 
yesterday we released new CLI and new Pro IDE
can you test `0.1.0` and let me know if it gets better for you?

thank you

per1234, 2021-02-18:
Hi all. Thanks so much for your valuable feedback,

Would you mind downloading [the new 0.1.4 release of Arduino Pro IDE](https://github.com/arduino/arduino-pro-ide/releases/tag/0.1.4) and letting us know whether the bug still occurs?

There have been some nice advancements since the time of your reports, so I think the issue might have been fixed along the way.

ubidefeo, 2021-02-19:
@per1234 as you can see on Aug 26 2020 I already asked @abionics to test v 0.1.0 and we never got a reply.
I will therefore close this issue.
Please, @abionics , let us know and reopen in case

closed 2021-02-19 by ubidefeo

ubidefeo, 2021-02-19:
thank you for reacting so quick, @abionics 🙏🏼